garymueller/ESP-131GPIO #2: Relay is reversed

CaffeinatedFunctionality, 2020-09-24:
Apparently my relay is in reverse. Turning it on in the app turns off the relay turning it off turns on the relay. Any idea how I could reverse this?

garymueller, 2020-09-24:
Glad to see someone else is finding the code useful.  The quick fix would be to change ESP-131GPIO.ino line 81 from

digitalWrite(GpioVector[GpioIndex], (data >= DigitalThreshold) ? **HIGH : LOW**);
to 
digitalWrite(GpioVector[GpioIndex], (data >= DigitalThreshold) ? **LOW : HIGH**);

I haven't tested this but it should work if it is reversed.  Are you using the same hardware as the readme?  The longer fix would be for me to add an if statement for a Low/High trigger and add a GUI option to allow you to set it.

garymueller, 2020-09-24:
You'll also want to change line 272 the same way (by swapping LOW and HIGH)

Line 81 handles the 131 traffic, Line 272 handles the app test

emiliobool, 2020-10-05:
Hi, I had the same issue, I swapped the values in several places and it worked fine, I left it on overnight turned "off" (all pins high) and this morning it didn't turn on anymore, I mean I can't make the pins go low anymore, sort off, I managed to measure it with a multimeter once but then it stopped working again. 

I don't know much about electronics so I don't know if I did something fundamentally wrong but just sharing just in case.

So what I mean with this is, if I go to the test page, I click any of the toggles then I refresh the toggles still appear off, the values don't stick and I'm not able to measure the pin going low either. So not sure if my gpios are dead or what, still investigating. 
 
I will try flashing the unmodified firmware again.

emiliobool, 2020-10-05:
I just tried with the default firmware, same thing, it won't change, it stays at the initial value though. I don't know if this if proof or signal that my gpios are bust, but I do see the problem, the version of the board I have, and probably OP does too, is a board without transistors, meaning I was just sinking 5v through the pins, while the recommended version has transistors in front. Both boards look almost identical except for the missing transistors, so this is something to consider for people coming from google.

emiliobool, 2020-10-05:
Ok, sorry for spamming I swear I'm not doing it on purpose, uploading the files again which I guess resetted my configuration, made the pins work again. But I don't understand why, I tried setting static IP back and it stopped working, if I disable static IP then it starts working again. I'm completely clueless as to why the static IP setting matters and why it used to work but not anymore. 

At this point this is a separate issue which is probably not related, so if this is not a simple answer and you are interested on investigating it further I can open a separate issue.

garymueller, 2020-10-05:
I'm sorry to hear that you are having trouble and appreciate the feedback.  I am having trouble recreating the issue on my side which makes it difficult to debug.  Im curious what relay board you are using?  That might explain the HIGH/LOW issue thats been observed.  When I developed the code I left in on for a few weeks and would test it periodically and didnt observe it freeze.

closed 2020-10-08 by garymueller