garymueller/ESP-131GPIO #4: How to control from XLights?

lubbertkramer, 2021-05-02:
Hi,

TLDR:
Support question is how to add this ESP with channel relay to Xlights and how to control the individual channels, for example how to turn on channel 1 and 3 but turn of channel 5 and 8

-------------
Last year i started using xlights for lights in combination with home assistant for some esp devices like fog and water pump sprayers. Doing the lights from xlights sequence and the fog /water from home assistent gives a mismatch in timing. 

The reason it’s esphome based is because it are small fog makers/water pumps that  I use it in the Lemax/Luville Christmas village my girlfriend every years build. Last year we tried to make a watershow in the village like it’s in a theme park nearby (https://youtu.be/dzQdW_HgRlA) also see the attached images from last year. 

From several facebooks groups I was pointed to this git to make those devices also available in Xlights so I can make a large sequence on time there with lights and other devices. We are now preparing the ideas for this year so that's why I was hoping you could point in how to add this to xlights or maybe even add it to the readme? :) 

[image]
[image]
[image]

garymueller, 2021-05-02:
Theres a couple different ways that you could set this up in xlights.  I'll provide some information on how I set mine up which may help you.

1. Hardware
I included a picture of my build below.  The build holds 1 relay and controller in a cable guard box.  This should be familiar to you with regards to the readme instruction wiring diagrams.  I actually have 6 of these boxes in total for 48 relay channels

[image]

2. Xlights Controller (controller tab)
I set up a E131 controller.  The important thing is that the configuration match how you configured the esp(s) with regards to universe options.  I used multicast so that I can drive all 6 boxes from this single controller.

[image]

3. Xlights models (display tab)
In my use case I used these to support AC christmas lights on mini trees.  You need to make sure that the minitree maps to the relay controller.  I did this by defining the startchannel as seen below.  Take note the the controller is assigned to the "Relays" controller and that the start channel is 1.

[image]

You also need to make sure the model only uses 1 channel.  To make the model only 1 channel wide,  I defined it as a single color.  In reality the channel will have a vale of 0-255 and you can set the "on" threshold within the firmware.  

[image]

I then repeated the process and defined the remaining minitrees.  The startchannel of each minitree was updated so that it incremented by 1.   In the below photo you see that the models channels increment by 1 (3331,3332,3333,...).  I actually defined my startchannel to start from the previous model which helped me move the models by just moving the first one.  Both methods will achieve the same result.  The key is that the channels increment sequentially.  There are other methods possible so long as they define a single channel per model and that you can control them as relays.

[image]

With all the models defined you can then define the sequence to your liking.  You can use the ac toolbar or you can use effects. 
 I typically use effects such as singlestrand and chase.  I am also able to import shared sequences onto the minitrees which is nice not to start from scratch.

I hope these instructions help. and appreciate the feedback on how the instructions can be made more clear.  I may include them in the readme as a result of this exchange.

lubbertkramer, 2021-05-03:
Thanks, will try to flash it tomorrow and give it a try first with only one device connected directly to ESP for On/Off. Will follow up

lubbertkramer, 2021-09-08:
Hi,

finally found the time to work on this for the Christmas project. I have flashed the D1 with your code with the libraries but it's never coming up with an ip adress. 

I have added a dedicated 2.4GHZ wifi network in your code

Did you have this problem also?

Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 330724          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28493   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1508  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 4020  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26128 )         - zeroed variables      (global, static) in RAM/HEAP 
De schets gebruikt 364745 bytes (38%)  programma-opslagruimte. Maximum is 958448 bytes.
Globale variabelen gebruiken 31656 bytes (38%) van het dynamisch geheugen. Resteren 50264 bytes voor lokale variabelen. Maximum is 81920 bytes.
esptool.py v3.0
Serial port /dev/cu.usbserial-2130
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a4:cf:12:dd:76:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.0s
Flash params set to 0x0340
Compressed 368896 bytes to 264921...
Writing at 0x00000000... (5 %)
Writing at 0x00004000... (11 %)
Writing at 0x00008000... (17 %)
Writing at 0x0000c000... (23 %)
Writing at 0x00010000... (29 %)
Writing at 0x00014000... (35 %)
Writing at 0x00018000... (41 %)
Writing at 0x0001c000... (47 %)
Writing at 0x00020000... (52 %)
Writing at 0x00024000... (58 %)
Writing at 0x00028000... (64 %)
Writing at 0x0002c000... (70 %)
Writing at 0x00030000... (76 %)
Writing at 0x00034000... (82 %)
Writing at 0x00038000... (88 %)
Writing at 0x0003c000... (94 %)
Writing at 0x00040000... (100 %)
Wrote 368896 bytes (264921 compressed) at 0x00000000 in 6.8 seconds (effective 436.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[SPIFFS] data    : /Users/lubbertkramer/Downloads/ESP-131GPIO-master/ESP-131GPIO/data
[SPIFFS] size    : 64
[SPIFFS] page    : 256
[SPIFFS] block   : 4096
/index.html
/favicon.png
skipping .DS_Store
[SPIFFS] upload  : /var/folders/p3/cw88qfvd0v3_4fjl7jkh94mc0000gn/T/arduino_build_382700/ESP-131GPIO.spiffs.bin
[SPIFFS] address  : 0xEB000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : /dev/cu.usbserial-2130
[SPIFFS] speed    : 921600
[SPIFFS] python   : /Users/lubbertkramer/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3
[SPIFFS] uploader : /Users/lubbertkramer/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/upload.py

esptool.py v3.0
Serial port /dev/cu.usbserial-2130
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a4:cf:12:dd:76:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 65536 bytes to 6593...
Writing at 0x000eb000... (100 %)
Wrote 65536 bytes (6593 compressed) at 0x000eb000 in 0.2 seconds (effective 3059.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

garymueller, 2021-09-08:
I haven't run into that issue.  Once I configure the ssid and password it would connect to my network and get assigned an ip.  You could also assign a static ip in the configuration if you want. 

I was told recently that the software was dependent on esp8266 v2.74 (see issue #5 ). perhaps that is causing an issue for you.

I dont understand your comment "I have added a dedicated 2.4GHZ wifi network in your code".  Does that mean you modified the network connection code?

lubbertkramer, 2021-09-09:
It worked, reverting back to Esp8266 v2.74 solved it and the device showed up on my network. I think it's a good thing to add to the readme that the build needs to be done with this version, 3.x is not working

What I meant with the 2.4GHZ wifi network was that I have a dedicated IoT network that shouldn't give problems. But the problem not connecting was with the board version not my wifi

[image]

Now I have the D1 connected to the relay and in the WebUI test it's working. The connected fog maker is going on / off. Now I want to add it to XLights but I'm stuck, I can't figure out with your earlier post on how to add this now to Xlights.

Could you elaborate a bit more on how to add it to Xlights and add it to a model so that every relay port is a model and then turn if on/off on sequence? 

I have added it to xlights like this now:
[image]

garymueller, 2021-09-10:
Great idea on adding the version information to the readme!  It was already added when the issue was first discovered here https://github.com/garymueller/ESP-131GPIO/commit/4aa1396819012927546c635f57b789821c7c9cf7

### Adding to xlights
I see that you already created a relay controller to work with the models which is great.  Now you just need to add the model such that it uses only 1 channel per model.  There are multiple ways to do this as I mentioned above.  Since you had issues with the setup described above, I'll explain another way you can do that.

1) go to the layout tab
2) click "Create new channel block"
3) Place the channel block within the scene
4) in the left pane set the "# channels" to 8
5) set "Start Channel" to the Relay controller that you created

6) go to the sequencer tab and create a new sequence
7) double click the channel block relay model
  * it expands to the 8 strands/relays that you setup previously
8) drop the desired effects onto the timeline to meet your requirements
* you can use the on effect for absolute on/off which should meet your needs
* you can also drag the single strand chase effect onto the top level channel block to have a chase effect across all your relays.
* try the different effects until you find what you want.

I hope this helps.  Let me know if your successful.

lubbertkramer, 2021-09-12:
Thanks, above solution worked for me I have got it working in xLights! I will now start to make the show as mentioned in my startpost, I will post the result here as a share and how I have used in the sequence :)

garymueller, 2021-09-12:
I'm really glad you got it to work for you.  I look forward to seeing the end resut of your project.  In the meantime I will mark this issue as closed for now. Thanks

closed 2021-09-12 by garymueller