garymueller/ESP-131GPIO #5: stuck in a loop

kflott, 2021-08-19:
Not sure what I'm doing wrong, after uploading the device just seems to be stuck in a loop rebooting with an exception epc1=0x40202e89.

Serial output below:
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0005a0d0
~ld
{"network":{"hostname":"esps-98237f","ssid":"MySSID","password":"MyPassword","static":false,"static_ip":"192.168.1.100","static_netmask":"255.255.255.0","static_gateway":"192.168.1.1","access_point":true},"E131":{"multicast":"true","universe":1,"channel_offset":0},"GPIO":{"digital":"true","digital_threshold":127,"digital_lowlevel":false}}

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x40202e89 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 0190
3ffffea0:  3ffe88b7 3ffef1f4 3ffef1e0 40202e89  
3ffffeb0:  3fff03e4 3ffef1e0 00000000 00000100  
3ffffec0:  feefeffe feefeffe feefeffe 3fff0c7c  
3ffffed0:  00000020 000000fc 3fff0004 40211b08  
3ffffee0:  3fffff20 00000000 3fff097c 000000f0  
3ffffef0:  feefeffe feefeffe 00000578 00000578  
3fffff00:  000000f0 3fff0004 3ffefffc 4020ff25  
3fffff10:  00000578 40218dd8 00000000 000003e8  
3fffff20:  3fff0a7c 00000000 00000000 00000000  
3fffff30:  00000000 00000000 00000000 3ffef48c  
3fffff40:  00000000 000b000f 00000000 3fff04b4  
3fffff50:  3ffef1e0 00000001 3ffef28c 3ffef404  
3fffff60:  00000000 3fff0004 3ffefffc 4020ffbe  
3fffff70:  00000000 3ffe8ca5 3ffef48c 3ffef404  
3fffff80:  3fffdad0 3ffe8ca5 00000001 40202ea4  
3fffff90:  3fffdad0 3ffe8ca5 3ffef28c 40204461  
3fffffa0:  feefeffe 00000000 3ffef3f0 4020e864  
3fffffb0:  feefeffe feefeffe 3ffe85e4 40100d81  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

garymueller, 2021-08-19:
The only reason it should get into a loop is if it is unable to connect to the wifi and access_point is false.  It will continue to reboot until it connects to wifi in that state.  However you debug output indicates that access_point is true so that shouldn't be your issue.

I would try to reflash the firmware and html and see if that corrects the issue.  Another thing you could try is to boot it without it being connected to serial debug.  I did come across an issue on another firmware where somehow connecting to serial caused an endless loop.  Please let me know if either of these fix your issue.

kflott, 2021-08-20:
Thanks for the suggestions.  Tried reflashing, also tried another device.  No luck as of yet.  I'll try again when I get some time.  

Decoding the exception gives me below:

Exception 0: Illegal instruction
PC: 0x40202fa5
EXCVADDR: 0x00000000

Decoding stack results
0x40219960: spiffs_impl::SPIFFSImpl::_check_cb(spiffs_check_type, spiffs_check_report, unsigned int, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 326
0x40100b10: malloc(size_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 821
0x40211959: uart_init(int, int, int, int, int, size_t, bool) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\uart.cpp line 714
0x402103d0: spiffs_impl::SPIFFSImpl::spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 343
0x402103a8: spiffs_impl::SPIFFSImpl::spiffs_hal_write(unsigned int, unsigned int, unsigned char const*) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 337
0x402103bc: spiffs_impl::SPIFFSImpl::spiffs_hal_erase(unsigned int, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 340
0x4020cb6c: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/HardwareSerial.h line 193
0x40202fc0: LoadConfig() at C:\Users\user\OneDrive\Arduino\ESP-131GPIO/ESP-131GPIO.ino line 114
0x402046dd: setup() at C:\Users\user\OneDrive\Arduino\ESP-131GPIO/ESP-131GPIO.ino line 52
0x4020f630: loop_wrapper() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 198

garymueller, 2021-08-20:
It looks like its failing when trying to mount the filesystem.  Did you create the file system and transfer the data files using the below instruction.

4) Then select the "ESP8266 sketch data upload" found under the tools menu. This will upload the data directory (html) to the arduino.

kflott, 2021-08-20:
Confirmed the file was present, even used a test sketch to read index.html back from the filesystem.  Thanks for your suggestions.
Maybe it's something specific to my environment, I might try to compile on another PC later.  Is there a compiled binary somewhere I could try flashing?

kflott, 2021-08-20:
SOLVED!

On a forum someone recommended reverting the ESP8266 library to 2.x.  Rolled it back from 3.02 to 2.74 and it compiled and ran!

garymueller, 2021-08-21:
Thats great news.  I just started to look at it this morning to try to solve your issue when I happened to see your comment.  I'm glad  you got it to work.

closed 2021-08-21 by garymueller