garymueller/ESP-131GPIO #9: (compatability) WeMos d1 mini pro issue

Tom-Neverwinter, 2021-11-17:
Sadly when programming the wemos d1 mini pro it seems to hang and never come back up after interacting with it over wifi.

garymueller, 2021-11-17:
Is there any debug information from the serial monitor that you can provide?  If compiling from older firmware please upgrade to the latest.  There was an issue that was resolved with issue #6 .

Tom-Neverwinter, 2021-11-18:
I was using the version from one update previous. I didn't realize there was a version released in October. 

I will flash this tomorrw.

garymueller, 2021-11-18:
Also note that the file system library was changed over recently from to littlefs (https://github.com/garymueller/ESP-131GPIO/commit/7200295ab9f2921290f3f47455105dca09d7d9eb).  Please make sure that you flash the correct filesystem (not SPIFFS ).

Tom-Neverwinter, 2021-11-18:
attempted flash: https://pastebin.com/AFM3JNQK

upload littlefs failed.

garymueller, 2021-11-18:
Thankyou for the feedback.  It's unclear to me if you got this running or if there still is an issue.  Are you saying that it was failing when the ssid contained spaces and resolved when spaces were omitted?

Tom-Neverwinter, 2021-11-18:
no it seems it fails when the ap name is changed. I next need to test and see if its any item that changes causes it to crash. 

stability seems fine inside my house. however the ultimate test as always is outside where it will be running. Which I am also curious if its disconnecting thus trying to change to ap mode and then crashes? 

I will report my findings.

garymueller, 2021-11-21:
I'm not sure why its not connecting to your wifi. Does your SSID / password contain special characters and/or spaces?  I'm wondering if its not saving your settings properly.

Thanks for the info on the mini pro pins.  I can add them for others

Tom-Neverwinter, 2021-11-22:
just seems to be a weird connection issue. I dont have anything special or outside the wifi spec. it also doesn't seem to connect even when on my table line of sight maybe 20ft from my wifi access point. which has 2.4ghz and 5ghz capabilities. 

I'm off today and tomorrow so I will investigate some more after the trees get fixed. [yay late night rj45 fabrication fails XD ]

garymueller, 2021-12-15:
Can I close this issue? Seems to be resolved as an issue with your network setup

closed 2021-12-15 by Tom-Neverwinter