garymueller/ESP-131GPIO #10: xlights output

surfpgh6, 2021-12-14:
Greetings,

Xlights does not turn off my esp8266 relay controller before running sequence! 

I have another single relay controller that I loaded last year and it works perfectly! It stays on when you do a hard reset which is good and when I run xlights and hit the bulb button it turns off until I run the sequence! I cant remember for the life of me which code I loaded onto it but is has the same web interface as your code! Cant find that code to replicate another one with 4 relays on it!

Your code is not turning off when I hit the light bulb on xlights! It works well otherwise!
I would like it to default to on when booted up and then have xlights turn it off like my WLED controllers do! 
Everything needs to go dark before my sequence starts!

If you have some insight it would be greatly appreciated!

surfpgh6, 2021-12-15:
If I turn the relays on from the Web interface and run xlights they turn off before sequence starts which is great!
I would like them to default to on like switching from the web interface!

surfpgh6, 2021-12-15:
I Figured it out!

garymueller, 2021-12-15:
That's great. I'm out of town right now and haven't had a chance to look into it. Can you describe what fixed your issue so that it might help others.

closed 2021-12-15 by garymueller