garymueller/ESP-131GPIO #11: ESPAsyncE131-master/RingBuf.h:50:77: error: extended character ” is not valid in an identifier

giroquoi, 2021-12-14:
Hi Garymueller, 

I use your code last year for my christmas light, it was great.
I want to had some more and i try to flash a new wemos D1 mini but this time  I have this return about buf :

In file included from C:\Users\jpbet\Documents\Arduino\libraries\ESPAsyncE131-master/ESPAsyncE131.h:37,
                 from C:\Users\jpbet\Documents\Arduino\ESP-131GPIO\ESP-131GPIO.ino:4:
C:\Users\jpbet\Documents\Arduino\libraries\ESPAsyncE131-master/RingBuf.h:50:18: error: extended character “ is not valid in an identifier
   50 |         #warning “This library only fully supports AVR and ESP8266 Boards.”
      |                  ^
C:\Users\jpbet\Documents\Arduino\libraries\ESPAsyncE131-master/RingBuf.h:50:77: error: extended character ” is not valid in an identifier
   50 |         #warning “This library only fully supports AVR and ESP8266 Boards.”
      |                                                                           ^
Plusieurs bibliothèque trouvées pour "ESPAsyncE131.h"
Utilisé : C:\Users\jpbet\Documents\Arduino\libraries\ESPAsyncE131-master
Non utilisé : C:\Users\jpbet\Documents\Arduino\libraries\ESP_Async_E1.31
exit status 1
Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini

Could you help me ?

Thanks a lot

//https://github.com/garymueller/ESP-131GPIO

#include <ESP8266WiFi.h>
#include <ESPAsyncE131.h>
#include <ESPAsyncWebServer.h>
#include <ESPAsyncDNSServer.h>
#include <LittleFS.h>
#include <ArduinoJson.h>
#include <vector>

using namespace std;

// ***** USER SETUP STUFF *****
String ssid = "Freedelapassion"; // replace with your SSID.
String password = "tokaavai"; // replace with your password.

String CONFIG_FILE = "Config.json";
DynamicJsonDocument Config(1024);
AsyncWebServer server(80);
AsyncDNSServer dnsServer;
ESPAsyncE131 e131;

//for Wemos D1 R1 pins are 16,5,4,14,12,13,0,2
//vector<int> GpioVector{16,5,4,14,12,13,0,2};
//for Wemos D1 R2 pins are 16,5,4,0,2,14,12,13
vector<int> GpioVector{16,5,4,0,2,14,12,13};
int DigitalOn = HIGH;
int DigitalOff = LOW;

//Forward Declarations
String WebReplace(const String& var);

bool LoadConfig();
void SaveConfig(AsyncWebServerRequest* request);

void InitGpio();
void InitWifi();
void Init131();
void InitWeb();

/*
 * Sets up the initial state and starts sockets
 */
void setup() {
  Serial.begin(115200);
  delay(500);
  Serial.println("==============================");
[... 316 lines cut ...]
    if(Config["GPIO"]["digital_lowlevel"].as<bool>())
      return "checked";
     else
      return "";

  //Relay Page
  } else if (var == "RELAYS") {
    String Relays = "";
    for(int i = 0; i < GpioVector.size(); ++i) {
      Relays += "<label>Relay "+String(i+1)+" ("+GpioVector[i]+")</label>";
      Relays += "  <label class=\"switch\">";
      Relays += "  <input type=\"checkbox\" ";
      if(digitalRead(GpioVector[i]) == DigitalOn) {
        Relays += "checked";
      }
      Relays += " onclick=\"fetch('SetRelay?relay="+String(i)+"&checked='+this.checked);\">";
      Relays += "  <span class=\"slider round\"></span>";
      Relays += "</label>";
      Relays += "<br><br>";
    }
    return Relays;
  }
 
  return var;
}

garymueller, 2021-12-15:
I think I saw this before and had to update to the latest libraries under manage libraries. Please let me know if this helps and if not please let me know your library versions

garymueller, 2021-12-15:
FYI. Heres the library versions that work for me that I think might be relevant to your issue

ESPAsyncE131 = 1.03
ArduinoJSN = 6.18.5
E131 = 1.0.0
esp8266-framework = 1.1.5

giroquoi, 2021-12-16:
Thanks a lot, it was so simple actually. It works now with updated libraries !

closed 2021-12-16 by giroquoi