garymueller/ESP-131GPIO #12: Web Page Not Responsive

kc5uvc, 2022-12-21:
First, thank you so much for sharing this.  I'm very excited about getting it working.

I have downloaded everything, installed all libraries and tools, compiled and uploaded without error. The device is pulling a DHCP address (x.x.x.51).  I can ping the address without problem.  However, I cannot get the web page to come up.  The browser (Chrome) just says it cannot be found.  Any suggestions on where to start?  The only thing I have changed in the config is the SSID and password, and I can see that it is connected to the router.

garymueller, 2022-12-21:
From your description it sounds like the firmware is successfully loaded and running, however is unable to serve the index.html file.  The index.html file is uploaded from under the tools menu.  Did you upload the data using the "ESP8266 LittleFS data upload" option uner the tools menu?  Thats the first thing that comes to my mind from your description.

kc5uvc, 2022-12-22:
Yes, I did successfully add the tool (after installing v1.8) and uploaded after flashing the firmware.  It shows upload progress and states the hash value is valid.  I simply click on the Little FS tool and it uploads, correct?  Is there anything else specific I need to do?

kc5uvc, 2022-12-22:
I don't know if this will help or not, but I just connected to the device in AP mode.  The SSID did come up with the MAC of the D1 Mini.  I was able to connect to the device and was issued an ip of 192.168.4.2.  When I try to connect to 192.168.4.1, nothing appears.  It's like the web services are not running at all.

kc5uvc, 2022-12-22:
Well, whatever it was, it was on my end.  I completely started over and it is working perfectly.  Thank you so much for your time.

closed 2022-12-22 by kc5uvc

kc5uvc, 2022-12-22:
Gary could you contact me offline?
***@***.***
(870) 897-9555
_______________________________________
Charles E Jones, BS, NRP, FP-C, CCP-C, TP-C
***@***.***